berndpfrommer/flir_spinnaker_ros2 #3: Issues

FranzAlbers, 2022-04-22:
Hello, 

I'm running into the following error with the driver and ROS2 Galactic:

`ros2 launch flir_spinnaker_ros2 chameleon.launch.py
[INFO] [launch]: All log files can be found below /home/rst/.ros/log/2022-04-22-16-00-24-523416-rst-leaf-588698
[INFO] [launch]: Default logging verbosity is set to INFO
[<launch.substitutions.launch_configuration.LaunchConfiguration object at 0x7f7a1ccdcc40>, '_']
[INFO] [camera_driver_node-1]: process started with pid [588700]
[camera_driver_node-1] [INFO] [1650636024.730163265] [chameleon_s]: debug: 0
[camera_driver_node-1] [INFO] [1650636024.730450522] [chameleon_s]:  serial: 18497292
[camera_driver_node-1] [INFO] [1650636024.736319797] [chameleon_s]: using default calibration URL
[camera_driver_node-1] [INFO] [1650636024.736406129] [chameleon_s]: camera calibration URL: file:///home/rst/.ros/camera_info/chameleon_s.yaml
[camera_driver_node-1] [ERROR] [1650636024.736527862] [camera_calibration_parsers]: Unable to open camera calibration file [/home/rst/.ros/camera_info/chameleon_s.yaml]
[camera_driver_node-1] [WARN] [1650636024.736546182] [chameleon_s]: Camera calibration file /home/rst/.ros/camera_info/chameleon_s.yaml not found
[camera_driver_node-1] [INFO] [1650636026.068137220] [chameleon_s]: using spinnaker lib version: 2.3.0.77
[camera_driver_node-1] [WARN] [1650636026.171879187] [chameleon_s]: overwriting bad param with default: parameter 'gain_auto' has invalid type: cannot declare a statically typed parameter with an uninitialized value
[camera_driver_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[camera_driver_node-1]   what():  parameter 'gain_auto' has invalid type: cannot declare a statically typed parameter with an uninitialized value
[ERROR] [camera_driver_node-1]: process has died [pid 588700, exit code -6, cmd '/home/rst/autoware/install/flir_spinnaker_ros2/lib/flir_spinnaker_ros2/camera_driver_node --ros-args -r __node:=chameleon_s --params-file /tmp/launch_params_c32vf40m --params-file /tmp/launch_params_3v2czpi0 -r ~/control:=/exposure_control/control']`

The issue seems to be related to [this change](https://github.com/ros2/rclcpp/blob/master/rclcpp/doc/notes_on_statically_typed_parameters.md) in ROS Galactic. The node runs, if I enable `dynamic_typing` for the `ni.descriptor` in the `createCameraParameters()` function. However, this would be just a quick fix.

Given that it is mentioned that this package runs on ROS Galactic, is there a fix known?

berndpfrommer, 2022-04-22:
Looking into it....

berndpfrommer, 2022-04-23:
First thanks for submitting this issue and pointing me to the relevant change which was spot on. I didn't realize that my robot was still on foxy, so all testing was on foxy, not galactic.
A fix has been pushed, could you please test? My flir cameras are on the robot which is in a crate right now...

FranzAlbers, 2022-04-26:
Thank you very much for the quick help, this error seems to be fixed now.

I have one additional issue since the upgrade to ROS2 Galactic: Camera images seem to not be published anymore. I don't receive the camera images anymore with ros2 topic echo or in rviz2, even though I've change the Reliablity Policy for the image subscriber and the Fixed Frame in rviz. 

The output of the camera driver node usually is something as follows:
```
[camera_driver_node-1] [INFO] [1650988403.276135123] [camera_front]: frame rate in: 21.013 Hz, out:0 Hz, drop: 0%
```

Do you have any hint how I could solve the issues?

FranzAlbers, 2022-04-27:
The outputs are attached below. The topic `/camera_front/image_raw` is advertised, but no messages seem to be published on that topic. The only messages I receive when running the camera node are some `ImageMetaData` messages on the topic `/camera_front/meta`.

```
ros2 node info /camera_front
/camera_front
  Subscribers:
    /exposure_control/control: camera_control_msgs_ros2/msg/CameraControl
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /camera_front/camera_info: sensor_msgs/msg/CameraInfo
    /camera_front/image_raw: sensor_msgs/msg/Image
    /camera_front/meta: image_meta_msgs_ros2/msg/ImageMetaData
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /camera_front/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /camera_front/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /camera_front/get_parameters: rcl_interfaces/srv/GetParameters
    /camera_front/list_parameters: rcl_interfaces/srv/ListParameters
    /camera_front/set_parameters: rcl_interfaces/srv/SetParameters
    /camera_front/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
    /set_camera_info: sensor_msgs/srv/SetCameraInfo
  Service Clients:

  Action Servers:

  Action Clients:
```

```
ros2 topic info --verbose /camera_front/image_raw
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: camera_front
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 2d.93.10.01.26.2a.4e.c6.02.41.6b.ff.00.00.19.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 1000000000 nanoseconds
  Deadline: 5000000000 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 10000000000 nanoseconds

Subscription count: 2

Node name: _ros2cli_8582
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: b7.37.10.01.64.7a.a7.09.fe.db.35.f7.00.00.08.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  Durability: VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Node name: rviz
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: d8.88.10.01.5d.35.98.cf.ff.c4.57.74.00.00.27.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

```

FranzAlbers, 2022-04-27:
Setting the always_publish parameter fixed it for me. 
For some reason, `pub_.getNumSubscribers() ` returned always 0, even though the image is displayed in rviz and rviz obviously has subscribed.

berndpfrommer, 2022-04-27:
Could this be related to this issue? 
https://github.com/ros-perception/image_common/issues/238
Also something similar mentioned here:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/README.md
Not clear when/if this bug has been fixed.
Notably, the always_publish flag was not introduced by me but by "John Pratt" whose github handle I can no longer find. At that point I should have asked what the purpose was, but I didn't. Probably to work around the bug you found.
I never hit this issue under Foxy when testing but apparently it's there. Either I'm not using the transport correctly and/or it is a fringe case that has to do with transport/OS/hardware. But testing for getNumSubscribers() before publishing is actually good practice and is done in many drivers so I don't understand why there aren't more issues filed regarding this.

berndpfrommer, 2022-04-27:
Hmm, maybe it's because I was fiddling around with the qos profile?
Can you try just with the default profile? So after this line:
``rmw_qos_profile_t qosProf = rmw_qos_profile_default;``
delete all the modifications that are made to qosProf?

berndpfrommer, 2022-04-27:
Thanks for trying that. Would you mind trying that fix and possibly filing a PR? The robot is still in the crate and I'm working on another project so testing for me is not easy.
I'm still a bit baffled that I never hit this problem when I was testing under foxy. Must be some other setting in the environment that triggers this bug.
I'd be perfectly happy with the count_subscribers() work around. I don't really like the "always_publish" flag because if defaulted to true it introduces additional overhead if there are no subscribers.

closed 2022-04-27 by berndpfrommer

FranzAlbers, 2022-04-27:
Sure, I've added a pull request and removed the parameter.

Thank you for your help!