berndpfrommer/flir_spinnaker_ros2 #4: Support for Flir GigE Blackfly S?

arvinddraffen, 2022-04-27:
I saw that you had tested this ROS 2 driver on Flir USB3 cameras, but would this driver also support Flir GigE Blackfly S cameras? I see the Flir SDK supports both USB3 and GigE, but wasn't sure if that extended to this ROS 2 driver as well. Was considering getting a Flir GigE camera to use, but ROS 2 driver support is a factor.

Thanks.

berndpfrommer, 2022-04-27:
This is really difficult to say. AFAIK there is nothing in the ROS2 driver that is specific to USB3 and the spinnaker SDK should handle it transparently. Should. In practice the chance it'll work out of the box is less than 50%.
Having that said if you have basic linux and ROS2 skills I promise to help you make it work. It should not be too hard.

kjanesch, 2022-05-16:
Hi Bernd et al, I have USB3 and GigE versions of the same Blackfly S (BFS-(U3|PGE)-23S3C), and the GigE one works effectively out of the box with this driver. With the camera configured (more below), you just point it to the Serial Number as you do with a USB3 camera, and you're set. 

I personally spun off separate config launch files to catch any GigE-specific settings in the Transport Layer Control group. Mainly, this line to enable jumbo frames from the camera per FLIR's recommendations:
`gev_scps_packet_size int "TransportLayerControl/GigEVision/GevSCPSPacketSize"`
and in the parameters defined in the (new) launch file, I added the corresponding (and set it to the same MTU as the network interface):
`"gev_scps_packet_size": 9000`

As far as *setting up* the camera's IP address, if you're using it on a network with DHCP it will probably work straight away. I have the camera connected directly to a computer, so I can't confirm that. For me it starts up with a Link-Local (`169.254.`...) address. From SpinView, you can "force" an IP that'll fit your machine's IP and netmask settings (you have to anyway to configure or view it there, a good sanity-check step). If you then leave the camera on and disconnect it within SpinView (or close SpinView) it'll retain its IP address and settings until you power it off, and you can then fire the node right up.

You can not have to do the above by setting a static ("persistent") camera IP in SpinView, in "Transport Layer Control">"GigE Vision". Check the box for "**Current IP Configuration Persistent IP**" first to enable it, then set your desired addresses under "**Persistent IP Address**", "**Persistent Subnet Mask**" and "**Persistent Gateway**". NOTE: these *look* like regular IPs, but to set them you have to enter the 32-bit integer representation of the IP address/mask. By hand/calculator: convert the IP octets from decimal to hex, then combine them and convert to a 32-bit integer, ex: `192.168.0.1` -> `0xC0A80001` -> `3232235521`.

The "Transport Layer Control">"GigE Vision" section of SpinView is also where you'll find that "SCPS Packet Size" setting, which you can change when not capturing frames, and verify it works in SpinView and without needing to spin up a custom launch file to get started, though it helps, and you'll probably want one anyway to specify your camera's serial number.

I got most of my tips from FLIR's support pages, namely the below, and if you've got other questions I can try to answer them.
https://www.flir.com/support-center/iis/machine-vision/knowledge-base/lost-ethernet-data-packets-on-linux-systems/
https://www.flir.com/support-center/iis/machine-vision/application-note/troubleshooting-image-consistency-errors/

berndpfrommer, 2022-05-17:
@kjanesch, many thanks for testing and the detailed steps on configuring the IP for the GigE cameras.
Can I take your post above and cut-and-paste it into the README?

berndpfrommer, 2022-05-17:
Also, would you mind filing a PR with an example file in the "config" directory, maybe called "blackfly_s_gige.cfg" that uses gev_scps_packet_size as per your post above? Most everybody will want to run jumbo frames and people just love known-working example config files.

kjanesch, 2022-05-18:
@berndpfrommer you're more than welcome to use it in the README. I'll put together a PR with the example config/launch file shortly.

berndpfrommer, 2022-05-31:
@kjanesch do you have a good config file by now? Looks like there is already demand for it #12

kjanesch, 2022-06-02:
@berndpfrommer I imagine you have a notification for it as well, but it's been added as PR #13

berndpfrommer, 2022-06-04:
closed by PR #13

closed 2022-06-04 by berndpfrommer