berndpfrommer/flir_spinnaker_ros2 #8: Can't build from source

pepeRossRobotics, 2022-05-17:
Hello, 
I manually cloned the repository along with the `flir_spinnaker_common` and the `camera_control_msgs_ros2`.

When I try to build, I get this error:
```
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver.cpp:84:13: error: no declaration matches ‘std::string flir_spinnaker_common::Driver::setInt(const string&, int, int*)’
   84 | std::string Driver::setInt(
      |             ^~~~~~
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver.cpp:84:13: note: no functions named ‘std::string flir_spinnaker_common::Driver::setInt(const string&, int, int*)’
In file included from /home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver.cpp:17:
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/include/flir_spinnaker_common/driver.h:29:7: note: ‘class flir_spinnaker_common::Driver’ defined here
   29 | class Driver
      |       ^~~~~~
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver_impl.cpp: In member function ‘std::string flir_spinnaker_common::DriverImpl::setInt(const string&, int, int*)’:
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver_impl.cpp:220:21: warning: overflow in conversion from ‘double’ to ‘int’ changes value from ‘+QNaN’ to ‘0’ [-Woverflow]
  220 |   *retVal = std::nan("");
      |             ~~~~~~~~^~~~
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver_impl.cpp: In instantiation of ‘std::string flir_spinnaker_common::set_parameter(const string&, T2, T2*, const Spinnaker::CameraPtr&, bool) [with T1 = Spinnaker::GenApi::CFloatPtr; T2 = int; std::string = std::__cxx11::basic_string<char>]’:
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver_impl.cpp:222:75:   required from here
/home/pepe/ross_robotics/mk4_ws/src/extrm_mk4/ccu/mk4_sensing/flir_spinnaker_common/src/driver_impl.cpp:194:21: warning: overflow in conversion from ‘double’ to ‘int’ changes value from ‘+QNaN’ to ‘0’ [-Woverflow]
  194 |   *retVal = std::nan("");
```

I fell that the `flir_spinnaker_common` repository is missing an update. could you confirm if it is up to date?

best,
Pepe

berndpfrommer, 2022-05-17:
Yes, something's broken. Working on it.

berndpfrommer, 2022-05-17:
@FranzAlbers could you please file a PR against the flir_spinnaker_common repo with the missing setInt() method added?

berndpfrommer, 2022-05-17:
@pepeRossRobotics I pushed a temporary fix until @FranzAlbers files a PR with a tested implementation.
Can you see if it compiles now for you? The driver should work so long as you don't set integer parameters (which may also work but has not been tested)

FranzAlbers, 2022-05-17:
Sorry for missing pushing the `setInt` method to the flir_spinnaker_common repo, @berndpfrommer! 

I can confirm that the driver compiles with your temporary fix and that setting integer parameters works.

Besides the templates for `set_invalid,` I used more or less exactly the same code, so I propose making this temporary fix a permanent one without another PR from my side.

berndpfrommer, 2022-05-17:
@FranzAlbers sounds good, thanks for the super fast reply!

closed 2022-05-17 by berndpfrommer