berndpfrommer/flir_spinnaker_ros2 #9: package failed: flir_spinnaker_ros2

sssxrf, 2022-05-20:
Hi,
I was reported an error while running `colcon build`.
[image]
Also, I installed the spinnaker SDK 2.6.0 successfully.
[image]
Could you please help me out? Thank you so much

berndpfrommer, 2022-05-20:
What's missing is the camera-info-manager package. There is a way to install all missing dependencies of a workspace, by running *at the top of your workspace*:
```
rosdep install --from-paths src --ignore-src -r -y
```
Could you please run that for me and let me know if that automatically installed the missing camerainfo-manager package?
If yes then I'll include that in the README instructions.

In case the above command fails you can install it manually:
```
sudo apt install ros-galactic-camera-info-manager
```
(or ros-foxy-...., whatever distro you are running on)
But please first try the rosdep line to see if that works. Thanks!

sssxrf, 2022-05-20:
The rosdep line works. Thank you!

berndpfrommer, 2022-05-20:
updated documentation

closed 2022-05-20 by berndpfrommer