berndpfrommer/flir_spinnaker_ros2 #10: install the driver on Windows10

sssxrf, 2022-05-24:
After I install ros2 successifully on Windows, I tried to install your driver. However, I could not run `wstool` despite I install it with 
`pip install -U rosdep rosinstall_generator wstool rosinstall`
[image]
The picture below shows that the system could not recognize `wstool`
[image]

Do you know how to run it on Win10? Thank you so much!

berndpfrommer, 2022-06-18:
Any progress on this? Can this issue be closed?

sssxrf, 2022-07-07:
Sorry for the late reply, I'm trying to build my own on Windows 10. But I could not add the FLIR dependencies correctly. I put all the spinnaker files in the 'include' in my packages. However, when I `colcon build  `it, it seems it could not read '.lib' file. Could you please give me some instructions?

sssxrf, 2022-07-11:
Yes, it works on Ubuntu. However, I want to do a Windows version. Anyway, thank you for your help.