berndpfrommer/flir_spinnaker_ros2 #12: error occurs while connecting to a Blackfly S (GigE)

sssxrf, 2022-05-31:
I've added `gev_scps_packet_size int "TransportLayerControl/GigEVision/GevSCPSPacketSize"` in the blackfly_s.cfg and `"gev_scps_packet_size": 9000` in blackfly_s.launch.py as shown below.
[image]
[image]
I also set the current persistent IP in spinview(actually it is default IP address).
[image]
Then I ran `ros2 launch flir_spinnaker_ros2 blackfly_s.launch.py camera_name:=blackfly_0 serial:="'20262963'"`, it showed the error:
[image]
I thought I might misunderstand some part of your Gige setting. Could you please tell me how to do it correctly?

berndpfrommer, 2022-05-31:
Looks like the problem is due to a missing space in the config file: rather than ``#add`` it should read ``# add``.
I just pushed a change that should make the config file parser accept your config file. Could you please go into the src/flir_spinnaker_ros2 directory and run a ``git pull`` there?

sssxrf, 2022-05-31:
It has a new error while `colcon build' after I `git pull` your new code.
[image]

berndpfrommer, 2022-05-31:
try installing e.g ``ros-galactic-ament-cmake-clang-format``

sssxrf, 2022-05-31:
I think it might connect to the camera. However, it said the images acquired are incomplete. Then I turn the spinview to see what happens. It just turns to black images(it works well with spinview's default settings)
[image]
[image]

The below picture shows my camera's parameters' settings in the launch file
[image]

berndpfrommer, 2022-05-31:
Just to get the facts straight: default settings (no jumbo frames) work in SpinView, correct?
Then what exactly breaks SpinView? Switching on jumbo frames? To fully exploit the camera you will want to get jumbo frames working first, in SpinView, forget about the ROS2 driver until you have that settled.
In SpinView:
- any error messages?
- can you fiddle around with other parameters to make it work, like lower the frame rate?
Also, check that your network card supports Jumbo frames. You may have to enable them there. Under Linux I run ``ifconfig -a`` to get the MTU. The MTU must be >=9000.
https://linuxconfig.org/how-to-enable-jumbo-frames-in-linux
You can also google for the same procedure on Windoze.

sssxrf, 2022-05-31:
It seems to get the full images after I set my MTU to 9000! Thank you so much! I really appreciate your help.
[image]

BTW, is there a way to visualize the images stream, like how to add the 'opencv' library in the work space?

berndpfrommer, 2022-06-01:
http://wiki.ros.org/rqt_image_view

There is a corresponding ros2 package as well.

berndpfrommer, 2022-06-04:
I believe this is now resolved.

closed 2022-06-04 by berndpfrommer