berndpfrommer/flir_spinnaker_ros2 #14: Build fails with Foxy

ajordan5, 2022-06-17:
I am using ROS2 Foxy with Ubuntu 20.04, but I am not able to build this 

```
Starting >>> camera_control_msgs_ros2
Starting >>> flir_spinnaker_common
Starting >>> image_meta_msgs_ros2
Finished <<< flir_spinnaker_common [0.35s]
Finished <<< camera_control_msgs_ros2 [0.68s]
Finished <<< image_meta_msgs_ros2 [0.68s]
Starting >>> flir_spinnaker_ros2
--- stderr: flir_spinnaker_ros2                             
/usr/bin/ld: libcamera_driver.so: undefined reference to `camera_info_manager::CameraInfoManager::CameraInfoManager(rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libcamera_driver.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::setAssociatedLibraryPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/usr/bin/ld: libcamera_driver.so: undefined reference to `class_loader::impl::AbstractMetaObjectBase::AbstractMetaObjectBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/camera_driver_node.dir/build.make:217: camera_driver_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/camera_driver_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< flir_spinnaker_ros2 [0.53s, exited with code 2]

```

I have verified that both the camera_info_manager and class_loader packages are installed and I am only sourcing ROS2. Any ideas on what might be wrong here?

I'm happy to help if there is any work needed to get Foxy up and running. 

It may be worth noting that I changed one thing in the source code. As shown below, I commented out the setting of the `dynamic_typing` flag as the Foxy version of rcl_interfaces doesn't have a dynamic_typing field under `ParameterDescriptor.msg` anymore.

```
static rcl_interfaces::msg::ParameterDescriptor make_desc(
  const std::string name, int type)
{
  rcl_interfaces::msg::ParameterDescriptor desc;
  desc.name = name;
  desc.type = type;
  desc.description = name;
  // if (has_dynamic_typing<rcl_interfaces::msg::ParameterDescriptor>()) {
  //   desc.dynamic_typing = true;
  // }
  return (desc);
}
```

berndpfrommer, 2022-06-18:
Thanks for the feedback with the dynamic_typing under foxy! I confirmed it is indeed not compiling. Looking at the code I have no idea how it was ever supposed to compile. Will look for a fix later tonight.
Not sure what's going on with the link errors you are seeing. I don't get those, i.e. if I remove the dynamic_typing bug as you did the code compiles fine.. It's weird that it would find the header files but then fail at the link stage. Can you clean your workspace and then retry? Also, check that your ROS environment variables are set up properly, reinstall the ROS camerainfo manager package.

ajordan5, 2022-06-18:
Sure, here is the output of `printenv | grep -i ROS`

```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
PWD=/home/tpool2/flir_spinnaker_ros2_ws
AMENT_PREFIX_PATH=/opt/ros/foxy
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
ROS_MASTER_URI=http://localhost:11311
ROS_HOSTNAME=localhost
LD_LIBRARY_PATH=/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/foxy/bin:/home/tpool2/.local/bin:/opt/spinnaker/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=foxy

```

ajordan5, 2022-06-18:
Okay, I finally got it to build and I wish I could clearly identify what got it to work. I cleaned my workspace by removing the install, build and log folders. I also removed all sourcing statements from my `.bashrc` , then opened a new terminal and sourced my foxy distro. It now builds successfully!

Now, it successfully starts up, but rejects any acquisition framerate that I give it. 

```
[camera_driver_node-1] [INFO] [1655556006.826315010] [chameleon_s]: setting AcquisitionControl/AcquisitionFrameRate to: 10
[camera_driver_node-1] [WARN] [1655556006.827436207] [chameleon_s]: setting AcquisitionControl/AcquisitionFrameRate failed: node AcquisitionControl/AcquisitionFrameRate not available!
```

(tried 100.0, 60, 30, 10 so far)

The driver then repeatedly gives this output:

```
[camera_driver_node-1] Image incomplete: Image data is incomplete. This could be caused by missing packet(s). For more information see https://www.flir.com/support-center/iis/machine-vision/application-note/troubleshooting-image-consistency-errors/
```

ajordan5, 2022-06-18:
Update: I also get 
```
[camera_driver_node-1] Image incomplete: Image chunk data is invalid.
```

berndpfrommer, 2022-06-18:
What model is it. Chameleon USB3, or are you trying to get another camera working?
I'm pretty sure chunk_mode is available. I did test that chameleon launch file successfully at some point. Are you on the latest firmware? That camera is pretty ancient (one of my favorites though) and the firmware has changed quite a bit over the years.

berndpfrommer, 2022-06-18:
If the driver rejects frame rate settings, try the same config with SpinView. Can you change the frame rate there? Is the 30.1Hz just upper limit or can you set it lower? Sometimes auto exposure settings affect the frame rate because to get good exposure you need a minimum shutter time, thus limiting the frame rate. Also make sure you are running on USB3 indeed (if it works in SpinView then you are good on that front).

berndpfrommer, 2022-06-18:
I pushed a fix for the compiler errors under foxy. Would you mind testing? 
https://github.com/berndpfrommer/flir_spinnaker_ros2/commit/2a898277ffec4cb400af6ce56ec9485d6df93de2
If this works I'd like to close this issue and let you open another one regarding the frame rate settings :)

ajordan5, 2022-06-20:
> I pushed a fix for the compiler errors under foxy. Would you mind testing? [2a89827](https://github.com/berndpfrommer/flir_spinnaker_ros2/commit/2a898277ffec4cb400af6ce56ec9485d6df93de2) If this works I'd like to close this issue and let you open another one regarding the frame rate settings :)

Just tested and it compiles :smile: . Still need to disable chunk settings, but its working well once I do that. There's a good chance I'm running on an old firmware. I'll look into finding out the firmware and open up another issue concerning framerates if the issue persists. Thank you for your help!

berndpfrommer, 2022-06-20:
Alright, thanks for testing! Closing this issue then, foxy is now officially supported.

closed 2022-06-20 by berndpfrommer