berndpfrommer/flir_spinnaker_ros2 #15: CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):

omair50, 2022-07-18:
Hi, I got the following error, could you plz update the user guide?
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find SPINNAKER (missing: SPINNAKER_INCLUDE_DIR SPINNAKER_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindSPINNAKER.cmake:56 (find_package_handle_standard_args)
  CMakeLists.txt:22 (find_package)

berndpfrommer, 2022-07-18:
Did you install the Spinnaker SDK? What version? Is this on Ubuntu? What OS version?

omair50, 2022-07-18:
I installed the Spinnaker SDK, and it compiles now. Thanks for quick reply!

berndpfrommer, 2022-07-18:
You are welcome.

closed 2022-07-18 by berndpfrommer