berndpfrommer/flir_spinnaker_ros2 #27: Problem with Camera synchronization

dariodenardi, 2022-12-17:
Hi,

First thank you for this repo. It's very helpful!
I have a problem with camera synchronization. My setup is: Ubuntu 22.04, spinnaker 2.6.0.157 and ROS2 Humble. Also I use Blackfly S (USB 3).

If I execute script:

```
ros2 launch flir_spinnaker_ros2 stereo_synced.launch.py
```

script doen't work and I haven't any images. My output is:

```
ros2 launch flir_spinnaker_ros2 stereo_synced.launch.py
[INFO] [launch]: All log files can be found below /home/dario/.ros/log/2022-12-17-22-56-22-822435-dario-Surface-Laptop-Go-18303
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [18317]
[component_container-1] [INFO] [1671314183.235042433] [stereo_camera_container]: Load Library: /home/dario/flir_spinnaker_ros2_ws/install/flir_spinnaker_ros2/lib/libcamera_driver.so
[component_container-1] [INFO] [1671314183.297592805] [stereo_camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<flir_spinnaker_ros2::CameraDriver>
[component_container-1] [INFO] [1671314183.297646854] [stereo_camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<flir_spinnaker_ros2::CameraDriver>
[component_container-1] [INFO] [1671314183.300694465] [cam_0]: debug: 0
[component_container-1] [INFO] [1671314183.300872434] [cam_0]:  serial: 21143314
[component_container-1] [INFO] [1671314183.302997519] [cam_0]: using default calibration URL
[component_container-1] [INFO] [1671314183.303036773] [cam_0]: camera calibration URL: file:///home/dario/.ros/camera_info/cam_0.yaml
[component_container-1] [ERROR] [1671314183.303085573] [camera_calibration_parsers]: Unable to open camera calibration file [/home/dario/.ros/camera_info/cam_0.yaml]
[component_container-1] [WARN] [1671314183.303105170] [cam_0]: Camera calibration file /home/dario/.ros/camera_info/cam_0.yaml not found
[component_container-1] [INFO] [1671314185.282664739] [cam_0]: using spinnaker lib version: 2.6.0.157
[component_container-1] [INFO] [1671314185.393720421] [cam_0]: found camera with serial number: 21143314
[component_container-1] [INFO] [1671314185.676715427] [cam_0]: setting AnalogControl/GainAuto to: Off
[component_container-1] [INFO] [1671314185.681554272] [cam_0]: setting AnalogControl/Gain to: 0
[component_container-1] [INFO] [1671314185.684925211] [cam_0]: setting DigitalIOControl/LineSelector to: Line2
[component_container-1] [INFO] [1671314185.686547024] [cam_0]: setting DigitalIOControl/V3_3Enable to: 0
[component_container-1] [INFO] [1671314185.688342316] [cam_0]: setting DigitalIOControl/LineSelector to: Line3
[component_container-1] [INFO] [1671314185.689559297] [cam_0]: setting DigitalIOControl/LineMode to: Input
[component_container-1] [INFO] [1671314185.691316835] [cam_0]: setting AcquisitionControl/ExposureAuto to: Off
[component_container-1] [INFO] [1671314185.695083369] [cam_0]: setting AcquisitionControl/ExposureTime to: 9000
[component_container-1] [INFO] [1671314185.696934764] [cam_0]: setting AcquisitionControl/TriggerSelector to: FrameStart
[component_container-1] [INFO] [1671314185.697662793] [cam_0]: setting AcquisitionControl/TriggerMode to: On
[component_container-1] [INFO] [1671314185.699155758] [cam_0]: setting AcquisitionControl/TriggerSource to: Line3
[component_container-1] [INFO] [1671314185.700618095] [cam_0]: setting AcquisitionControl/TriggerDelay to: 11
[component_container-1] [INFO] [1671314185.702122324] [cam_0]: setting AcquisitionControl/TriggerOverlap to: ReadOut
[component_container-1] [INFO] [1671314185.703710791] [cam_0]: setting ChunkDataControl/ChunkModeActive to: 1
[component_container-1] [INFO] [1671314185.706380892] [cam_0]: setting ChunkDataControl/ChunkSelector to: FrameID
[component_container-1] [INFO] [1671314185.707322291] [cam_0]: setting ChunkDataControl/ChunkEnable to: 1
[component_container-1] [INFO] [1671314185.709682561] [cam_0]: setting ChunkDataControl/ChunkSelector to: ExposureTime
[component_container-1] [INFO] [1671314185.710031774] [cam_0]: setting ChunkDataControl/ChunkEnable to: 1
[component_container-1] [INFO] [1671314185.712294696] [cam_0]: setting ChunkDataControl/ChunkSelector to: Gain
[component_container-1] [INFO] [1671314185.712830496] [cam_0]: setting ChunkDataControl/ChunkEnable to: 1
[component_container-1] [INFO] [1671314185.715080677] [cam_0]: setting ChunkDataControl/ChunkSelector to: Timestamp
[component_container-1] [INFO] [1671314185.715511403] [cam_0]: setting ChunkDataControl/ChunkEnable to: 1
[component_container-1] [INFO] [1671314185.725996926] [cam_0]: camera has pixel format: BayerRG8
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/cam_0' in container '/stereo_camera_container'
[component_container-1] [INFO] [1671314185.728005583] [stereo_camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<flir_spinnaker_ros2::CameraDriver>
[component_container-1] [INFO] [1671314185.728041936] [stereo_camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<flir_spinnaker_ros2::CameraDriver>
[component_container-1] [INFO] [1671314185.730759775] [cam_1]: debug: 0
[component_container-1] [INFO] [1671314185.730926035] [cam_1]:  serial: 21143313
[component_container-1] [INFO] [1671314185.731824374] [cam_1]: using default calibration URL
[component_container-1] [INFO] [1671314185.731855989] [cam_1]: camera calibration URL: file:///home/dario/.ros/camera_info/cam_1.yaml
[component_container-1] [ERROR] [1671314185.731883858] [camera_calibration_parsers]: Unable to open camera calibration file [/home/dario/.ros/camera_info/cam_1.yaml]
[component_container-1] [WARN] [1671314185.731887899] [cam_1]: Camera calibration file /home/dario/.ros/camera_info/cam_1.yaml not found
[component_container-1] [INFO] [1671314185.745909518] [cam_1]: using spinnaker lib version: 2.6.0.157
[component_container-1] [INFO] [1671314185.765768060] [cam_1]: found camera with serial number: 21143313
[component_container-1] [INFO] [1671314186.001346677] [cam_1]: setting AnalogControl/GainAuto to: Off
[component_container-1] [INFO] [1671314186.004789521] [cam_1]: setting AnalogControl/Gain to: 0
[component_container-1] [INFO] [1671314186.006584991] [cam_1]: setting DigitalIOControl/LineSelector to: Line2
[component_container-1] [INFO] [1671314186.007540293] [cam_1]: setting DigitalIOControl/V3_3Enable to: 0
[component_container-1] [INFO] [1671314186.008701918] [cam_1]: setting DigitalIOControl/LineSelector to: Line3
[component_container-1] [INFO] [1671314186.009245310] [cam_1]: setting DigitalIOControl/LineMode to: Input
[component_container-1] [INFO] [1671314186.010272664] [cam_1]: setting AcquisitionControl/ExposureAuto to: Off
[... 182 lines cut ...]
    }

def generate_launch_description():
    """Launch camera node."""
    flir_dir = get_package_share_directory('flir_spinnaker_ros2')
    config_dir = flir_dir + '/config/'
    name_arg = LaunchArg('camera_name', default_value='blackfly_s',
                         description='camera name')
    serial_arg = LaunchArg('serial', default_value="'21143314'",
                           description='serial number')

    node = Node(package='flir_spinnaker_ros2',
                executable='camera_driver_node',
                output='screen',
                name=[LaunchConfig('camera_name')],
                parameters=[camera_params,
                            {'parameter_file': config_dir + 'blackfly_s.cfg',
                             # 'dump_node_map': True,
                             'serial_number': [LaunchConfig('serial')]}],
                remappings=[('~/control', '/exposure_control/control'), ])

    return LaunchDescription([name_arg, serial_arg, node])
```

Am I doing something wrong or can you find a solution to start everything in ```stereo_synced.launch.py```?

berndpfrommer, 2022-12-17:
I have not tested on that combination of Ubuntu/spinnaker but last time I tested on Ubuntu 18.04 with that stereo_synced.launch.py file, it definitely worked.
Here are a couple of debugging steps:
- from the stereo script, comment out the launch of the second camera. Does the single one then run?
- There are several USB3 memory limits that need to be bumped for Linux per FLIR instructions. Maybe some kernel resource prevents it from running when started in a single address space (composable node). Often the installer already bumps USB3 memory limits but just worthwhile double checking. Also, any suspicious messages in /var/log/syslog when you try to start the cameras?
- can you get the cameras to operate free running (switch off external trigger) when launched in the stereo script? So use the settings from the free running original blackfly_s.launch.py as it is checked into the repo, copy them into the stereo launch file and see if you can get the two cameras to produce frames w/o sync.

dariodenardi, 2022-12-18:
I did a lot of tests and I have tried with another pc with the same configuration (ubuntu 22.04, spinnaker 2.6.0.157 ...).

I answer by points:

1) No, it doesn't work. The problem is this parameter:

```
'trigger_mode': 'On',
```

if I change "on" to "off" script works. but what does disabling this feature mean? Is wired hardware sync disabled?

2) I don't see anything strange:

```
Dec 18 16:09:39 dario-Modern-14-B11MOL kernel: [ 2460.729497] usb 4-2: USB disconnect, device number 8
Dec 18 16:09:41 dario-Modern-14-B11MOL kernel: [ 2462.082975] usb 4-1: USB disconnect, device number 7
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.204039] usb 4-1: new SuperSpeed USB device number 9 using xhci_hcd
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.227510] usb 4-1: New USB device found, idVendor=1e10, idProduct=4000, bcdDevice= 0.00
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.227516] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.227519] usb 4-1: Product: Blackfly S BFS-U3-32S4C
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.227521] usb 4-1: Manufacturer: FLIR
Dec 18 16:09:49 dario-Modern-14-B11MOL kernel: [ 2470.227522] usb 4-1: SerialNumber: 01429F11
Dec 18 16:09:49 dario-Modern-14-B11MOL mtp-probe: checking bus 4, device 9: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1"
Dec 18 16:09:49 dario-Modern-14-B11MOL mtp-probe: bus: 4, device: 9 was not an MTP device
Dec 18 16:09:49 dario-Modern-14-B11MOL mtp-probe: checking bus 4, device 9: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1"
Dec 18 16:09:49 dario-Modern-14-B11MOL mtp-probe: bus: 4, device: 9 was not an MTP device
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.352343] usb 4-2: new SuperSpeed USB device number 10 using xhci_hcd
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.374927] usb 4-2: New USB device found, idVendor=1e10, idProduct=4000, bcdDevice= 0.00
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.374934] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.374936] usb 4-2: Product: Blackfly S BFS-U3-32S4C
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.374938] usb 4-2: Manufacturer: FLIR
Dec 18 16:09:50 dario-Modern-14-B11MOL kernel: [ 2471.374939] usb 4-2: SerialNumber: 01429F12
Dec 18 16:09:50 dario-Modern-14-B11MOL mtp-probe: checking bus 4, device 10: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2"
Dec 18 16:09:50 dario-Modern-14-B11MOL mtp-probe: bus: 4, device: 10 was not an MTP device
Dec 18 16:09:50 dario-Modern-14-B11MOL mtp-probe: checking bus 4, device 10: "/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2"
Dec 18 16:09:50 dario-Modern-14-B11MOL mtp-probe: bus: 4, device: 10 was not an MTP device
Dec 18 16:09:52 dario-Modern-14-B11MOL dbus-daemon[1798]: [session uid=1000 pid=1798] Activating service name='org.gnome.gedit' requested by ':1.224' (uid=1000 pid=7830 comm="/usr/bin/nautilus --gapplication-service " label="unconfined")
Dec 18 16:09:52 dario-Modern-14-B11MOL dbus-daemon[1798]: [session uid=1000 pid=1798] Successfully activated service 'org.gnome.gedit'
Dec 18 16:09:52 dario-Modern-14-B11MOL gnome-shell[2005]: meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
```

3) If I use the parameters of ```blackfly_s.launch.py```, ```stereo_synced.launch.py``` works

what do you suggest me?

berndpfrommer, 2022-12-18:
If you switch "trigger_mode" to off then the camera is free running, i.e. it will produce frames without a trigger signal (hardware sync).
I just checked and saw that you edited your original post. You originally claimed that it worked with hardware sync if you ran as separate notes, but then edited that claim away. That's a big difference. I only got your original post via email, not the edit. So is it then working with hardware sync if you run as individual nodes?
My feeling is that your hardware sync signal is not working, and that's why you get frames from the stereo node when you switch trigger_mode to "off".
So next use SpinView to configure the camera for hardware sync (trigger mode on), just for a single node. Does the camera produce frames when the hardware sync is connected? Does it stop producing frames when disconnected? Then you know your hardware sync is working.
Next, bring up two cameras in SpinView, and try to get them working with your hardware sync signal. Again make 100% sure that your hardware sync signal is really working by disconnecting it to make sure the sync signal is working.
And BTW, the stereo_synced.launch.py assumes you have an external trigger signal. Is that what you have? Or are you trying to run one as master (produces the signal) and one as slave (receives the signal). Then you need a different setup (and also must have the correct cable/pull up resistor).

dariodenardi, 2022-12-20:
I seem to understand that trigger connection is different than master-slave connection.

On the flir site I can only find a schematic. Can you confirm that I need to recreate these connections here?

https://www.flir.eu/support-center/iis/machine-vision/application-note/configuring-synchronized-capture-with-multiple-cameras/

berndpfrommer, 2022-12-20:
So do you want master-slave (nowadays called primary-secondary) or do you have an external trigger source (like e.g. an IMU)?

For primary-secondary setup for 2 Blackfly_S cameras, create a sync cable following the link you posted. Here is the relevant diagram:
[image]

dariodenardi, 2022-12-20:
yes, I want create a primary-secondary setup. I've already created the connections.

reading the link I think that I must change params. Also I create two different ```camera_params```, one for primary camera and one for secondary camera, right?

```camera_params = {
    'debug': False,
    'compute_brightness': True,
    'dump_node_map': False,
    'gain_auto': 'Off',
    'gain': 0,
    'exposure_auto': 'Off',
    'exposure_time': 9000,
    'line2_selector': 'Line2',
    'line2_v33enable': False,
    'line3_selector': 'Line3',
    'line3_linemode': 'Input',
    'trigger_selector': 'FrameStart',
    'trigger_mode': 'On',
    'trigger_source': 'Line3',
    'trigger_delay': 9,
    'trigger_overlap': 'ReadOut',
    'chunk_mode_active': True,
    'chunk_selector_frame_id': 'FrameID',
    'chunk_enable_frame_id': True,
    'chunk_selector_exposure_time': 'ExposureTime',
    'chunk_enable_exposure_time': True,
    'chunk_selector_gain': 'Gain',
    'chunk_enable_gain': True,
    'chunk_selector_timestamp': 'Timestamp',
    'chunk_enable_timestamp': True,
    }```

berndpfrommer, 2022-12-20:
Correct. Primary and secondary need different parameters. On the primary you need to enable the 3.3V power output as well (above line2_v33enable needs to be set to 'True').
First use SpinView to get the hardware triggering to work. Once that works, adopt the same settings in the ROS driver.

dariodenardi, 2022-12-21:
Thank you. My problem was the terminology. In documentation, the word "trigger" is used in general and I thought there was only one type of trigger.

I change the params in this way:

```
camera_params1 = {
    'debug': False,
    'compute_brightness': True,
    'dump_node_map': False,
    'gain_auto': 'Off',
    'gain': 0,
    'exposure_auto': 'Off',
    'exposure_time': 9000,
    'line1_selector': 'Line1',
    'line1_linemode': 'Output',
    'line2_selector': 'Line2',
    'line2_v33enable': True,
    'trigger_mode': 'Off',
    'chunk_mode_active': True,
    'chunk_selector_frame_id': 'FrameID',
    'chunk_enable_frame_id': True,
    'chunk_selector_exposure_time': 'ExposureTime',
    'chunk_enable_exposure_time': True,
    'chunk_selector_gain': 'Gain',
    'chunk_enable_gain': True,
    'chunk_selector_timestamp': 'Timestamp',
    'chunk_enable_timestamp': True,
    }
```

```
camera_params2 = {
    'debug': False,
    'compute_brightness': True,
    'dump_node_map': False,
    'gain_auto': 'Off',
    'gain': 0,
    'exposure_auto': 'Off',
    'exposure_time': 9000,
    'trigger_selector': 'FrameStart',
    'trigger_mode': 'On',
    'trigger_source': 'Line3',
    'trigger_delay': 9,
    'trigger_overlap': 'ReadOut',
    'chunk_mode_active': True,
    'chunk_selector_frame_id': 'FrameID',
    'chunk_enable_frame_id': True,
    'chunk_selector_exposure_time': 'ExposureTime',
    'chunk_enable_exposure_time': True,
    'chunk_selector_gain': 'Gain',
    'chunk_enable_gain': True,
    'chunk_selector_timestamp': 'Timestamp',
    'chunk_enable_timestamp': True,
    }
```

outputs line up:

[image]

berndpfrommer, 2023-01-05:
Closing this issue as it seems to be resolved.

closed 2023-01-05 by berndpfrommer