berndpfrommer/flir_spinnaker_ros2 #28: Framerate drop issues

Thomas-Merlet, 2023-01-05:
Hi,

I'm running into issues while testing the driver in ROS2.

I'm able to compile and launch the driver without issues, correctly connecting to a USB3 or a GiGE FLIR Blackfly BFLY-U3-23S6C. I'm trying to get the image at 20Hz.

However, I've encountered the following issues:
- When launching the driver and having one subscriber (like `rqt_image_view`), everything works fine and the image seems to be at the required framerate. However, when launching another subscriber like `ros2 topic hz`, we can see a significant framerate drop in the image view and the reported frequency is about 5Hz. The flir_spinnaker_ros2 driver outputs drop at about 90%, from 20Hz to 5Hz. . When killing the other subscriber, everything goes back to normal. Note that this issue is not always present, and can sometimes appear during runtime. I've encountered the same problem with USB3 or GiGE cameras.

- When only launching the driver, everything works fine and the image seems to be at the required framerate. However, when launching multiple other ROS1 and ROS2 nodes, I can see a significant framerate drop in the image view, but the flir_spinnaker_ros2 output indicates no issues even though the image is not smooth.
The camera is directly connected to the computer through USB3. Note that I've tried using the following ROS1 driver https://github.com/ros-drivers/flir_camera_driver, and everything's working fine even if multiple nodes are running.  

**OS**: Ubuntu 20.04 LTS
**ROS**: ROS2 Galactic
**Spinnaker**: 3.0.0.118
**flir_spinnaker_ros2**: [63e17e1](https://github.com/berndpfrommer/flir_spinnaker_ros2/commit/63e17e1b793b99ff0fa87d8f5009a3a041f09549)

Thanks for your help

berndpfrommer, 2023-01-05:
I didn't quite understand this part of your comment: "The flir_spinnaker_ros2 driver outputs drop at about 90%, from 20Hz to 5Hz.". Could you please post a sample of the driver's log to the console?

Without the driver log it's hard to say what's going on but from what you write it feels like this is a ROS2 issue more than a driver issue. A couple of other data points that would be interesting to know:
- Is everything (driver, viewer etc) running on the same host or are you going across the network for any of the subscribers?
- What hardware are you running this on (raspberry pi, laptop, or big fat PC)?
- Could you check the CPU load of driver and subscribers while you experience the frame drops?
- Is the host running out of CPU resources (i.e. all cores busy)?

A warning about ros2 topic hz: it is not an accurate tool for measuring frequency because its code is so inefficient that it may drop data itself. Check CPU load of that process to see that it is not maxing out on CPU. To be fair at 20hz it should not be a problem, but just something to be aware of.
The camera driver itself is actually fairly simple. It takes an image from the camera and publishes it. Unless it fails to call to publish(), the problem is in ROS2 land, or the way the driver is using it, i.e. the quality-of-service profile used:
https://github.com/berndpfrommer/flir_spinnaker_ros2/blob/63e17e1b793b99ff0fa87d8f5009a3a041f09549/src/camera_driver.cpp#L580
Feel free to experiment with "image_queue_size". You can also try to just use the default profile by commenting out the lines that modify the default profile (following the line linked above).
The following suggestion will *not* help you with viewer issues, but may be interesting for other subscribers in your setup. You can run the the driver as a composable node in the same address space as the consumer. This way you get better performance and bypass the middleware. The stereo launch script gives an example of how to do that: https://github.com/berndpfrommer/flir_spinnaker_ros2/blob/master/launch/stereo_synced.launch.py
Lastly, maybe the nuclear option of changing the middleware will make a difference: https://docs.ros.org/en/foxy/How-To-Guides/Working-with-multiple-RMW-implementations.html

berndpfrommer, 2023-01-05:
And another thing: you are the first one to report a blackfly camera working. Would you mind posting your config file (or submit a pull request). This way people will know that the driver works for that camera as well and they have a config file to start with.

Thomas-Merlet, 2023-01-17:
Many thanks for your detailled explanation. 

- Everything is running on the host (driver, viewer ect)
- We are running that on high specs PC (I7, RTX 3070, 32GB RAM)
- CPU load of driver and subscribers is normal (about 30% at most)
- The host doesn't seem to be running out of CPU resources

Here is the ouptut of the driver when using the default configuration and using multiple subscribers as ROS2 nodes:
```
[camera_driver_node-1] [INFO] [1673626296.412305334] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:4.95549 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626301.442486003] [camera_front_center_tele]: frame rate in: 20.0081 Hz, out:5.16883 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626306.471832320] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:5.16964 Hz, drop: 296%
[camera_driver_node-1] [INFO] [1673626311.509092495] [camera_front_center_tele]: frame rate in: 20.0087 Hz, out:5.16155 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626316.540432215] [camera_front_center_tele]: frame rate in: 20.0083 Hz, out:4.96883 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626321.575739543] [camera_front_center_tele]: frame rate in: 20.0088 Hz, out:5.16358 Hz, drop: 284.615%
[camera_driver_node-1] [INFO] [1673626326.415927242] [camera_front_center_tele]: frame rate in: 20.0087 Hz, out:4.95845 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626331.445957968] [camera_front_center_tele]: frame rate in: 20.0107 Hz, out:5.16895 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626336.484135379] [camera_front_center_tele]: frame rate in: 20.0108 Hz, out:5.16059 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626341.512418516] [camera_front_center_tele]: frame rate in: 20.0088 Hz, out:4.97188 Hz, drop: 296%
[camera_driver_node-1] [INFO] [1673626346.549061657] [camera_front_center_tele]: frame rate in: 20.0088 Hz, out:4.96363 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626351.579308780] [camera_front_center_tele]: frame rate in: 20.0083 Hz, out:5.16873 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626356.421895946] [camera_front_center_tele]: frame rate in: 20.009 Hz, out:5.16253 Hz, drop: 288%
[camera_driver_node-1] [INFO] [1673626361.453958466] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:4.96814 Hz, drop: 296%
[camera_driver_node-1] [INFO] [1673626366.484383825] [camera_front_center_tele]: frame rate in: 20.0092 Hz, out:4.96976 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626371.523109438] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:5.16006 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626376.554924052] [camera_front_center_tele]: frame rate in: 20.0086 Hz, out:5.1671 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626381.390647718] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:4.96307 Hz, drop: 295.833%
[camera_driver_node-1] [INFO] [1673626386.427591036] [camera_front_center_tele]: frame rate in: 20.0085 Hz, out:4.96332 Hz, drop: 300%
[camera_driver_node-1] [INFO] [1673626391.459119221] [camera_front_center_tele]: frame rate in: 20.0088 Hz, out:5.16743 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626396.495255580] [camera_front_center_tele]: frame rate in: 20.0088 Hz, out:5.1627 Hz, drop: 288.462%
[camera_driver_node-1] [INFO] [1673626401.526636468] [camera_front_center_tele]: frame rate in: 20.0084 Hz, out:5.16757 Hz, drop: 284.615%
[camera_driver_node-1] [INFO] [1673626406.561349746] [camera_front_center_tele]: frame rate in: 20.009 Hz, out:5.16411 Hz, drop: 300%
```

We've made other tests based on your recommandations and here are the results:

- **Changing the quality-of-service of the publisher**: We didn't noticed any difference changing that and the queue size. I believe a queue size of 1 is a better configuration to avoid latency so we've kept that value. 

- **Using the same component container**: We've develop an equivalent of ros2 topic hz as a composable node to mesure frequency, and used the stereo_synced.launch.py as you suggested. Using that, we were able to get a stable frequency of 20Hz, even if multiples other nodes are running which is a good news. 

- **Changing the DDS RMW implementation**: We've tried switching to Fast DDS and noticed a difference in the publication rate of the driver, with less frequency drop and a more stable framerate. The ros2 topic hz command with a window size of 10 messages was giving a frequency of about 17Hz (we are targeting 20Hz), compared to about 6Hz with cyclonedds. However, the driver output reports no error in both cases:
```
[camera_driver_node-1] [INFO] [1673627081.618850429] [camera_front_center_tele]: frame rate in: 20.0094 Hz, out:19.9993 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627086.618668812] [camera_front_center_tele]: frame rate in: 20.0082 Hz, out:20.0007 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627091.618660521] [camera_front_center_tele]: frame rate in: 20.0084 Hz, out:20 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627096.618658904] [camera_front_center_tele]: frame rate in: 20.0079 Hz, out:20 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627101.618663978] [camera_front_center_tele]: frame rate in: 20.0084 Hz, out:20 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627106.618664585] [camera_front_center_tele]: frame rate in: 20.0089 Hz, out:20 Hz, drop: 0%
[camera_driver_node-1] [INFO] [1673627111.618662074] [camera_front_center_tele]: frame rate in: 20.0085 Hz, out:20 Hz, drop: 0%
[... cut ...]
[camera_driver_node-1] [INFO] [1673627166.618656901] [camera_front_center_tele]: frame rate in: 20.0082 Hz, out:20 Hz, drop: 0%
```
I beleive we have to use a composable node if we want to get good results. 

We are struggling to get a smooth visualization of the image as rqt_image_view is not implement as a composable node. 
As you said, we believe based on those results that the issue is more on related to ROS2 than to the driver code itself. 

For the config file, we've used the blackfly_s_gige.launch.py and just changed the serial number.

berndpfrommer, 2023-01-17:
After seeing the driver output I must back paddle on my claims about this being a ROS2 issue.
Obviously the drop rate > 100% is nonsense. I fixed this with PR #29. However the driver is not as simple as I thought: it has the outgoing ROS2 queue (managed by the image_queue_size parameter) but also an internal queue (buffer queue) that I had forgotten about. From all I can tell it was that queue that was dropping the frames. I added a new parameter ("buffer_queue_size") so you can now configure it. Please try and see if that makes a difference. The hard coded default size before was 2 images.
Now the question is why that queue fills up in your case and not for me. My laptop and can run 5 viewers (rqt_image_view) and 1 rostopic hz (which reports the right frequency btw) with my blackfly S at 66Hz. Zero drops, and all viewers look smooth. Each viewer consumes about 70%CPU. This is with ROS2 galactic running the default middleware (I presume cyclone).
The fact that you are seeing drops in the buffer queue means that the call to publish() via the ROS2 framework takes an unreasonable amount of time. This feels like a setup issue. Do you have another host that you can run on, with a completely fresh ROS2 installation?

berndpfrommer, 2023-02-11:
I'm hitting strange issues too, but only when the network is congested. These issues are completely in ROS2 land. I raised this as a question [here](https://answers.ros.org/question/412409/how-to-get-asynchronous-publish-working-on-ros2-galactic-with-fastrtps/).

berndpfrommer, 2023-05-25:
Closing this issue because it seems to be a ROS2 middleware problem that cannot be addressed by the driver. Please re-open if new leads about the root cause of this.

closed 2023-05-25 by berndpfrommer