berndpfrommer/flir_spinnaker_ros2 #32: calibration problem

dariodenardi, 2023-02-05:
Hi,

I have two cameras BlackFly S. I calibrated them and I put the file into the default path. When I run two cameras the files of calibration are read by ROS. In fact, when I write in a new terminal:

ros2 topic echo /cam_0/camera_info

I can see the values. But the images of the camera are published without considering these values. Is there a way to do this or do I have to do it myself?

berndpfrommer, 2023-02-05:
Sorry, but what do you mean with "not considering the value". All you can
expect is the driver to publish camera info and image.

dariodenardi, 2023-02-05:
it seems to me that it doesn't work. there can be any value in the file but the image is always the same.

berndpfrommer, 2023-02-06:
The driver will not rectify or modify the image in any form. Downstream
software can use the camera info to modify the image as needed.

dariodenardi, 2023-02-06:
Thank you for the explanation

closed 2023-02-06 by dariodenardi