berndpfrommer/flir_spinnaker_ros2 #33: switch camera format

AndreV84, 2023-02-09:
how could we use type different than
[bayer_rggb8] ?
Thanks

berndpfrommer, 2023-02-09:
Please pull and have a look at the blackfly_s launch file for hints on how to change the pixel format.
I have tested the subset of pixel formats that are both supported by ROS and the BlackFly S. If you need other formats: look at PR #34 and add the formats you need to both the flir_spinnaker_common and flir_spinnaker_ros2.

AndreV84, 2023-02-14:
@berndpfrommer Could you extend on the following, please?
from devs:
" why the /camera_info topic is not publishing the actual resolution that camera is set to"?
Thanks
AV

berndpfrommer, 2023-02-14:
Were you able to switch pixel formats successfully?

AndreV84, 2023-02-15:
yes, it was implemented somehow by folks to switch the format
thank you

AndreV84, 2023-02-20:
@berndpfrommer could you point out how could we tune exposure parameters of the camera? Thanks

berndpfrommer, 2023-02-20:
The ROS2 driver itself itself is just a thin wrapper that supports close to nothing by default.
You will need to add ROS2 parameters yourself by editing the config files.
This is a multi-step process:
1) Go into SpinView and find the exact names of the fields you want to tune.
2) Map the SpinView fields to a ROS2 parameter via the config file for your camera. For instance the "exposure_time" field has already been mapped in the config file for the blackfly S:
https://github.com/berndpfrommer/flir_spinnaker_ros2/blob/master/config/blackfly_s.cfg
Any other things you want to tune, you add them there.
3) Pass the parameters to the driver in the launch file (see blackfly_s launch file for example)

Once the driver has started, you can run "ros2 run rqt_reconfigure rqt_reconfigure" to change the parameters on the fly.