berndpfrommer/flir_spinnaker_ros2 #36: Image and camera_info topic not in sync

spirosperos, 2023-02-21:
I am trying to run apriltag detection using the [ros2 apriltag library](https://github.com/christianrauch/apriltag_ros).
However I am experiencing this warning:

> Topics '/blackfly_0/image_raw' and '/camera_info' do not appear to be synchronized. In the last 10s:
> 	Image messages received:      3
> 	CameraInfo messages received: 7
> 	Synchronized pairs:           2

How can I sync the camera info and the image topic? Thank you.

berndpfrommer, 2023-02-21:
There's nothing you should have to do, they should be synchronized because the same header stamp is put on when they are published.

But bugs are always an option so let's directly check if the time stamps are good. What do you get when you run simultaneously in two separate terminals:
 
`ros2 topic echo /blackfly_s/camera_info | grep -A 2 stamp`
and
`ros2 topic echo /blackfly_s/image_raw | grep -A 2 stamp`

For me the time stamps line up:

From camera_info:
```
--
  stamp:
    sec: 1677006792
    nanosec: 705401249
```

And from image_raw:
```
--                                           
  stamp:                                     
    sec: 1677006792                          
    nanosec: 705401249   
```

spirosperos, 2023-02-23:
For me they aren't lining up:

[image]

berndpfrommer, 2023-02-23:
They won't arrive at the identical time of course so your left and right column might be shifted by one or two slots. If you shift your image_raw (right) column up by one slot then the time stamps are identical.

berndpfrommer, 2023-02-23:
I should have looked at your original message more carefully.:

```
The Topics '/blackfly_0/image_raw' and '/camera_info' do not appear to be synchronized. In the last 10s:
Image messages received: 3
CameraInfo messages received: 7
Synchronized pairs: 2
```
It looks like your image_raw messages are being dropped and your camera_info messages make it through (they are smaller and consume less bandwidth). Are you running across wifi?

https://answers.ros.org/question/48071/image_raw-and-camera_info-not-synchronized-in-wlan/

closed 2023-03-17 by berndpfrommer