berndpfrommer/flir_spinnaker_ros2 #38: Image incomplete: Image has missing packets.

kairhe, 2023-03-16:
When I started the Launch file, I had the following problems, but I do n’t understand how to solve it. Can you help me provide an idea? thank you very much!
[Error1.txt](https://github.com/berndpfrommer/flir_spinnaker_ros2/files/10988156/Error1.txt)

kairhe, 2023-03-20:
> First thing: start SpinView (part of the spinnaker SDK). Make sure you get full fps using SpinView, only then move on to the ROS2 driver. Is this a GigE camera? If yes, enable jumbo packets. Also check that your [network interface is configured for jumbo frames](https://www.howtouselinux.com/post/check-mtu-size-in-linux) (large MTU).

Your advice is useful! I had resolved it under your help.Now i can use this driver without any problem.
By the way,I want to say that i tested this driver in Ubuntu 22.04 with Ros2-foxy on the Gige camera .

berndpfrommer, 2023-04-06:
Thanks for the feedback. Closing this issue as solved.

closed 2023-04-06 by berndpfrommer