berndpfrommer/flir_spinnaker_ros2 #45: No image in Nvidia Jetson Xavier

danicannt, 2023-05-10:
Hi,

He have been trying to configure the node in a Nvidia Jetson Xavier. We've installed the SDK for the arm64 architecture and so. When we try to open the spinview app the camera does not shows anything. We have it working in an Nvidia TX2 and in different typologies of systems, but it does not work in this one. 

Do you know if there is any known issue in this systems?

berndpfrommer, 2023-05-10:
I have a Blackfly S working on the Xavier AGX just fine.
Couple of trouble shooting tips:
- google for USB flir memory limit linux or something like that. You may need to bump kernel parameters
- Run "lsusb -t" to check that the camera is showing up on the USB bus, and that it is running as USB3 (5000Mbits), not USB2 (480Mbits).
- Connect the camera *with the same cable* to a different host to make sure it works there
- There is a wonderful command in linux called "strace". It shows all system calls. You run it with e.g. "strace -f SpinView >& out.txt". Then look at out.txt for clues if SpinView cannot open a file due to permissions, or doesn't find a file. The output can be overwhelming, but often you can get a clue what's going wrong.
-

berndpfrommer, 2023-05-24:
Closing because no response.

closed 2023-05-24 by berndpfrommer