berndpfrommer/flir_spinnaker_ros2 #48: 'Spinnaker::Exception' error while trying to acquire footage from FLIR Blackfly Camera

emrebzkurt, 2023-06-15:
I am using Ubuntu 20.04 focal on a NVIDIA arm64 device. Currently I have Spinnaker version 2.7.0.128 and ROS2 Foxy. The code worked fine couple of days ago while testing but now the acquisition suddenly stops after couple of seconds by itself. I have also tried it with different devices and ROS2 Galactic as well but got the same result.

```shell
[camera_driver_node-1] [INFO] [1686830427.224923861] [blackfly_0]: rate [Hz] in  20.00 out   0.00 drop   0%
[camera_driver_node-1] [INFO] [1686830432.224712821] [blackfly_0]: rate [Hz] in  20.00 out   0.00 drop   0%
[camera_driver_node-1] WARNING: acquisition timeout, restarting!
[camera_driver_node-1] terminate called after throwing an instance of 'Spinnaker::Exception'
[camera_driver_node-1]   what():  Spinnaker: Camera is not started. [-1002]
[ERROR] [camera_driver_node-1]: process has died [pid 3299, exit code -6, cmd '/home/ymir/flir_spinnaker_ros2_ws/src/install/flir_spinnaker_ros2/lib/flir_spinnaker_ros2/camera_driver_node --ros-args -r __node:=blackfly_0 --params-file /tmp/launch_params_urah7csx --params-file /tmp/launch_params_v41hg6ww -r ~/control:=/exposure_control/control'].
```

I am using this command to execute the launch file and the `camera_params` are the default of the repo.

berndpfrommer, 2023-06-15:
Apparently the camera does not produce images for more than 3 seconds, which causes an acquisition timeout. Do you have any idea why that is happening?
In normal operation I did not see those timeouts very often, which is why this situation is poorly debugged, so the driver crashes. There are comments related to that in the driver_impl.cpp in the flir_spinnaker_common repository.
Does the SpinView tool work just fine, for a prolonged period of time?

emrebzkurt, 2023-06-16:
I have no idea why the image acquisition stops. I have tested it with two different FLIR cameras, one with 5320x4600 width and height and another with a loser resolution to check whether there is an acquisition problem but they gave the same result.
SpinView doesn't woek as well. I am unable to get a proper video output from the SpinView. I can only get single frames when I constantly press the zoom in/out buttons.

emrebzkurt, 2023-06-16:
Seems like the issue is related to a USB connection problem between the Camera and the NVIDIA device. I have watched `lsusb` and observed that the camera randomly disappears suggesting that the connection broke down. I am closing this issue since it isn't a software related problem.

closed 2023-06-16 by emrebzkurt

berndpfrommer, 2023-06-16:
I noticed that the FLIR cameras are sensitive to the quality of the USB3 cable being used. I usually buy the cable from FLIR despite the higher cost. Make sure you have the USB kernel parameters set per FLIR instructions to rule out config settings as the source of the disconnects. Maybe your nvidia device has several different USB3 ports, that's worth exploring. Also consider putting a USB3 hub inbetween (works for me on Xavier AGX + 2 x blackfly S).